Carbon225/raspberry-sbus #2: Bad timing with outgoing SBUS packets

sophierahn, 2020-10-29:
I'm trying to feed SBUS instructions from a Jetson Nano into an F4 flight controller using betaflight. Betaflight itsn't detecting any sort of output. 

I used a logic analyzer to check the output and the output rate is super low, but when I adjust the example code send_to_self.cpp to play with timing (the current method only triggers every second) the data occasionally sends twice back-to-back. 

This could be an issue with betaflight but the output rate from this library is also an issue. Has there been a successful experience with trying to control a flight controller with this library?

Carbon225, 2020-10-29:
I use this library to receive messages from an Frsky receiver to control a ROS robot. I then send packets to an ESP32-based custom FC.

I tested timings with an FTDI adapter doing simple packet passthrough (RX->FC) and I did have pretty bad "timings". What I mean by that is that packets had a major delay and were less frequent. That is caused by the bad realtime behavior of FTDI adapters. However using the UART on the Pi the packets are retransmitted almost immediately and the outgoing stream perfectly reassembles the input.

I am quite certain this library meets the SBUS spec and should work with any FC. If you could post some waveforms from the analyzer maybe I could help.

Carbon225, 2020-10-29:
@sophierahn I just added a passthrough example to the repo and tried it with my normal Betaflight drone. On a Raspberry Pi 3 it works just fine and the drone can fly. Here are some waveforms (bottom - RX, top - data sent to FC).
[image]

sophierahn, 2020-10-30:
[image]

This the output vs input I'm seeing using your new passthrough example (thanks btw). 

As you can see, its not sending packets as often as its receiving and sometimes the Jetson is sending out double packets. Your placement of the write command logically should have it sending just as often but for some reason it isn't.

My wiring is a bit long. I've got the X8R connected to physical signal converter, then into the RX pin on a USB-tty converter. The TX pin of that same converter is passing through another signal converter and then into the flight controller. The data rate going in looks good, making me think that the converter shouldn't be an issue, but I am also new to programming at this level.

sophierahn, 2020-10-30:
I found that adding a 15 millisecond delay in the loop eliminated the double packets but did not fix the rate issue. Any smaller delay than that had no effect, any larger delay still eliminated the double packet but also decreased the write frequency. 

I'm realizing what I just said about the data rate going in being fine was not helpful as that data was taken from the the receiver itself before it had gone through the converter. By removing the "if(now>lastprint) in onPacket, allowing the input to print with every loop, the rate picked up. I think this still confirms the usb to tty converter isn't slowing things down. Is this a faulty assumption. This didn't change anything with the output.

Carbon225, 2020-10-30:
That looks like typical buffering behavior. I am not sure if it's the input or the output but I'd guess that the write command queues bytes and the UART adapter sends them later. So sometimes it sends fine but sometimes it "misses" a packet and sends it later.
I'm pretty sure that's exactly what I experienced with my FTDI adapter.

https://github.com/projectgus/hairless-midiserial
In this MIDI to serial project there is a feature that improves realtime performance of FTDI chips. When using an FTDI it asks you for administrator permissions and does some magic that actually makes things better. When I have time I might investigate more and implement the same thing but for now I don't know what they are doing exactly.

I would recommend using the UART on the jetson directly (assuming it has one) or trying different USB adapters. If you can tolerate the lower packet frequency you could also limit the rate at which you send packets.

I also think it should work quite well even the way it is now. All packets get transmitted only with a small delay which could be an issue for racing but it should be fine if the drone has stabilization turned on. This is actually what I'm doing in one of my projects. I have betaflight doing all the PIDs and I just send it angles so I don't have to worry about stabilization and can focus on automation.

Carbon225, 2020-10-30:
Actually, here's the file:
https://github.com/projectgus/hairless-midiserial/blob/master/src/PortLatency_linux.cpp

It seems they just set a single flag when configuring the serial port. If you want you could try modifying my setup code to do the same thing. I might try this when I have some time.

Carbon225, 2020-10-30:
"By removing the "if(now>lastprint) in onPacket, allowing the input to print with every loop, the rate picked up" 

I don't understand, can you explain more?

sophierahn, 2020-10-30:
Wow, everything works perfectly now in betaflight using the passthrough example. I don't understand why send_to_self wasn't working. write() was moved to onPacket instead of being called within the read() loop, but I would think onPacket should be called as frequently as read(). If there is a glaring difference between the two examples I'm missing, could you explain it to me? 

In response to your last comment, I was messing around with how often the print statements were called. The current method of only allowing the packets to print to terminal when now > lastPrint only allows one print every second (that seems to be the resolution of the variable type), so setting the if statement to if(true) let the packet print much more frequently. This mostly proved to me that the FTDI was letting data through faster than one packet per second. It didn't necessarily confirm there wasn't a slow down. 

Thank you so so sososososos much for your help. I was losing my mind a bit there and you fixed a big problem on my agenda. I really appreciate your help.

Carbon225, 2020-10-30:
Glad to hear it.
As for the difference between send_to_self and passthrough there is a major one.
The read() in passthrough is a blocking function. So when it is called the kernel puts the process to sleep and wakes it up when a packet arrives. If you looked at the CPU usage it should be minimal.

However, in the send_to_self example read() does not block. It checks if new data arrived and returns immediately. This allows you to easily send and receive packets from the same loop (same thread). This leads to high CPU usage which is why I implemented the blocking version.

The proper solution is to run the blocking read on a separate thread. It is more complicated which is why I have the non-blocking version (it's like programming an Arduino). Here's an example, my ROS SBUS library:
https://github.com/Carbon225/ros-sbus

closed 2020-10-31 by Carbon225