Carbon225/raspberry-sbus #3: Doesn't compile on Raspberry Pi 4.

Dimcon, 2021-05-09:
I've tried everything. Even ```#undef termios``` to no avail.
``` C++
In file included from ../src/sbus_driver.c:8:
In file included from /usr/include/arm-linux-gnueabihf/asm/termbits.h:1:
/usr/include/asm-generic/termbits.h:12:8: error: redefinition of 'termios'
struct termios {
       ^
/usr/include/arm-linux-gnueabihf/bits/termios.h:28:8: note: previous definition is here
struct termios
       ^
In file included from demo.cpp:9:
../src/sbus_driver.c:26:14: error: cannot initialize a variable of type 'uint8_t *' (aka 'unsigned char *') with an rvalue
      of type 'const uint8_t *' (aka 'const unsigned char *')
    uint8_t *payload = packet + 1;
             ^         ~~~~~~~~~~
2 errors generated.
```

Carbon225, 2021-05-09:
This seems to be caused by some very strict compiler options. The line with the payload pointer just needs a pointer to const to please the compiler (I should definitely change that).
Now the stuff with the termios struct is weird because it seems to be completely outside of the source file. As in the error is because of some conflicting includes. I will try figuring it out tomorrow.
Generally, if you want to just get it running try setting some lighter compiler options or even an older C standard.

Carbon225, 2021-05-10:
I guess I never made it clear but to use the library you just have to git clone and then do
```cmake
add_subdirectory(path/to/raspberry-sbus)
target_link_libraries(your_target PUBLIC sbuslib)
```
in your `CMakeLists.txt`. Then you can just include the headers.
I will add that to the readme later.

Dimcon, 2021-05-10:
Aaah.
Thanks for having a look for me!
I'll give it a go with those changes in CMakeLists. I cloned the repo and just imported both sbus_driver.c and SBUS.cpp into my demo.cpp main, and then compiled them all together with g++ demo.cpp. I later tried clang demo.cpp with -I./include -I/src with similar issues. 

I'll let you know how it goes 😃

closed 2021-07-22 by Carbon225