Carbon225/raspberry-sbus #10: How to do the connection from Rasberry Pi 4 to sbus?

research-boy, 2021-08-05:
I am new to this , might feel like a silly question :).
This is what i understood, connect the sbus tx to the Rasbeery Pi 4  GPIO14 .
Is there any need to use inverter in between the above two ?
What i am trying to do is to control blackmagic studio camera settings  through sbus , which i need to send control signals to appropriate channels in the sbus.
Will be a great help if you can guide me how to test it out with what you have made.
Thanks

Carbon225, 2021-08-05:
If you want to test the library connect RX to TX (gpio 14 to 15) and run the send_to_self example.
Most drones use active high signals that's why I mention using an inverter but I don't know about cameras. That's actually quite awesome as I never imagined cameras had SBUS.

Carbon225, 2021-08-05:
Just to clarify, you're supposed to connect gpio 14 which is TX to the RX pin on the camera.
Also do the raspberry pi setup from the readme (device tree overlays) to activate that UART.

research-boy, 2021-08-05:
@Carbon225 Thanks for quick replay, how normally people use is use a futaba remote to send data to this, which works fine. What i want know is whether it is sending  the same signal as a futaba ?

Carbon225, 2021-08-05:
It works for sending just fine. It will send an active low signal which is normal for UART. So for a drone you would need an inverter.

research-boy, 2021-08-05:
Thanks, will update you

research-boy, 2021-08-05:
@Carbon225 Hi , I would like to know is there any way i can invert the signal in the code itself

Carbon225, 2021-08-05:
I have not implemented that. I think it might be possible with certain adapters but I do not know much about it. If you know how to invert the signal in software please let me know, I will add it. I do not think many adapters support this feature though.

Carbon225, 2021-08-05:
Right now I'm working on the v2.1 version and I have a fully functional setup on a breadboard so I can give you schematics.

research-boy, 2021-08-05:
@Carbon225 cool , will be great if u can share ASAP , About inverting , this is something i found :
https://github.com/Cleric-K/vJoySerialFeeder/blob/master/Arduino/Sbus_invert/Sbus_invert.ino
https://github.com/Cleric-K/vJoySerialFeeder/blob/master/Docs/Sbus.md

Carbon225, 2021-08-05:
I had this in the readme: https://electronicspost.com/wp-content/uploads/2015/11/Transistor-as-an-inverter.png
I have a level converter as well to boost the signal to 5V but I guess your camera might not need that. And even most 5V electronics work with 3.3V signals.

Carbon225, 2021-08-05:
[image]
This is exactly what I have right now. Additionally I have that level converter right after the raspberry TX pin.

Carbon225, 2021-08-05:
This kind of level converter: https://www.sparkfun.com/products/12009
Though like I said it might not be necessary.

Carbon225, 2021-08-05:
Those links you sent are for an arduino inverter I think. It's quite cute. I don't know if that's what you wanted (a seperate device to work as an inverter) or if you wanted software to invert the signal directly on the sender. I think it was the latter.

research-boy, 2021-08-05:
I think you can use IC74HC04 for inverting, i am also not sure about the Arduino code i shared.

Carbon225, 2021-08-05:
This is a quote from that repo:
`
SBUS protocol sends data over a UART pin configured at 100 000 bps and using 8-E-2 framing. It also uses inverted signal levels. This makes it impossible for most USB UART converters to read the signal directly - the signal must first be inverted.
`
So like I thought most adapters cannot invert the signal. Even most microcontrollers don't have that feature. The ESP32 can do it I think. But this library is for computers so I'd want it to be as compatible as possible.

Carbon225, 2021-08-05:
I do not know that IC but I guess there might be inverters like that. If you're designing a PCB you might want to use it. But really a single transistor is all you need.

Carbon225, 2021-08-05:
I don't know if you've ever done any kind of electronics projects (I mean with some analog stuff and the like) so I understand if you're afraid of it. If you're careful it shouldn't be a problem.

research-boy, 2021-08-05:
@Carbon225 I understand to read will need inverter , but to write it didn't make sense. :)

Carbon225, 2021-08-05:
Do you know if your camera needs an active high signal (inverted)? Like in a drone.

research-boy, 2021-08-05:
@Carbon225 i couldn't find anything in the documents , but i tried using a futaba transmitter and reciever , which worked.
The reciever i have is https://futabausa.com/wp-content/uploads/2018/09/R2008SB.pdf
If you want know about the camera control : https://documents.blackmagicdesign.com/DeveloperManuals/BlackmagicCameraControl.pdf?_v=1539929000000

Carbon225, 2021-08-05:
So yes it seems you will need an inverter for TX. I don't have any send-only examples but really it's just the sbus.write() function and you can adapt the send_to_self example. You can make a loop that sends a packet every 10ms which is similar to real hobby receivers like your futaba. Make sure to use a delay. Any faster and your camera most likely won't recognize the packets.
I'm quite interested how you manage because it's such an original application.

closed 2024-01-27 by Carbon225