Carbon225/raspberry-sbus #13: Readme - aditional uarts on rpi4

gabrielbeu-ifsc, 2023-02-15:
> To use additional UARTs only on Raspberry Pi 4
> add dtoverlay=uartX to /boot/config.txt where X is 2, 3, 4 or 5 to enable another UART
> reboot and use /dev/ttyAMAX where X is the UART you chose

In case of using UART1 and another, for example UART3, the ttyAMAx is numbered sequentially. 
So, in this case, ttyAMA0 is for UART0 and ttyAMA1 is for UART3.

Carbon225, 2023-02-15:
That sounds correct, thanks.

gabrielbeu-ifsc, 2023-02-15:
You are welcome.

Thanks for the SBUS code on RPI.

Prof. Gabriel Macedo, M.e Eng.

DAE - Departamento Acadêmico de Eletrotécnica

Câmpus Florianópolis - Av. Mauro Ramos, 950 - Centro

Instituto Federal de Educação, Ciência e Tecnologia de Santa Catarina

Carbon225, 2023-02-16:
Fixed

closed 2023-02-16 by Carbon225