Carbon225/raspberry-sbus #16: Raspberry Pi SBUS Radio for betaflight fc

roy-Rik99, 2024-10-24:
Hi, I have gone through your code and it is really good. I want to use it for my uav where I want my Raspberry Pi to act as a radio controller for my flight controller. Is it possible to use your library for this task? If yes, can you please post a demo code to do so?

I am using a Raspberry Pi Zero 2W and my flight controller is STMF405. Your help will be greatly appreciated! Thanks in advance!

Carbon225, 2024-10-24:
Here's the relevant example: https://github.com/Carbon225/raspberry-sbus/blob/master/examples/send_to_self.cpp

It does both sending and receiving (if you want to hook up a radio receiver to the raspberry).

If you need just sending then you only have to `.install` and `.write`

```cpp
SBUS sbus;
sbus.install(ttyPath.c_str(), true); // false work too if you need non-blocking receiving

sbus_packet_t packet = {0};
// populate packet channels

// make sure to limit sending frequency, SBUS is not that fast
sbus.write(packet);
```

Carbon225, 2024-10-24:
To enable the uart try raspi-config and "enable console output on serial console -> no" and "enable uart hardware -> yes".

Or follow my readme "use built-in UART on Raspberry Pi 3/4" because the zero 2 is basically a pi 3.

roy-Rik99, 2024-10-24:
Thanks a ton bro!
I would like to contribute to your code if possible! Let me know if there are any loopholes you haven't addressed yet.

Actually I want to control my uav over 4G/5G. So, I can get unlimited range.

I am already using mavlink but every fc requires a radio other it switches to failsafe mode.

roy-Rik99, 2024-10-24:
> Here's the relevant example: https://github.com/Carbon225/raspberry-sbus/blob/master/examples/send_to_self.cpp
>
>
>
> It does both sending and receiving (if you want to hook up a radio receiver to the [...]

Is there any set packet data format?

Carbon225, 2024-10-24:
This is the struct:

```cpp
struct sbus_packet_t
{
    uint16_t channels[16];
    bool ch17, ch18;
    bool failsafe;
    bool frameLost;
};
```

Usually you just set the 16 channels.
 
I'm working on a rewrite in https://github.com/Carbon225/raspberry-sbus/pull/15

It's using std::function for callbacks which makes it easy to pass class functions as callbacks.

And I added full TBS Crossfire support. I've been flying on that PR branch with CRSF and a crazy fast racer drone for a few months now and I'll merge once I update the readme. I totally recommend CRSF over SBUS if your hardware supports it (no UART inversion is needed). SBUS is still supported of course.

Carbon225, 2024-10-24:
CRSF is bidirectional so you even get telemetry like battery status, gps, attitude, ...

roy-Rik99, 2024-10-24:
I have been using Ardupilot with a apm2.8 since my college days..Mission-Planner seemed to be a nightmare to me🥲
It's been over 6 years of tinkering with it and it is still a bit of a pain to use.

I recently got a stm32f4 board and I am currently tinkering with it. So, I switched to Betaflight. Loved the UI! Way simpler and intuitive than Mission Planner.

My goal is gps guided missions and  switching to manual control when needed. I also have a csi camera with my pi for fpv control when gps is unavailable.

I wanted to use csrf but then again I have to create a library for it so I was thinking about MAVLINK..

I switched back to ardupilot and mission planner since my fc supported it but it's been a nightmare of a experience but hopefully I'll get through it.

roy-Rik99, 2024-10-24:
Lucky to meet a fellow drone racer! Where you live if you don't mind me asking?